yaqwsx/Pinion #11: offset between the image and the overlay

domhardt, 2021-06-06:
There is an offset between the image and the overlay in Safari 14.1.1 on MacOS 10.15.7
[image]

dzarda, 2021-06-07:
Also the case on main w/ Windows/Firefox 90 *when maximized*:
[image]

OndroNR, 2021-06-08:
Chrome 91, macOS 11.4, retina display

[image]

yaqwsx, 2021-06-08:
This was probably broken by 92f28cd3c8137e4b274be2adc954b06ba8d48d2e, should be fixed in 7e802695fd610745050dc3acfefc9ea5b74c76da. Please let me know if it works. I don't have Mac available to test it.

closed 2021-07-25 by yaqwsx