yaqwsx/Pinion #13: TH component not clickable from other side

dzarda, 2021-06-07:
I feel like we should allow having a throughhole component clickable (highlight visible) from both sides.

Suggested logic for driving the clickableness from both sides: 

```
isSMD = are all pads SMD
clickableFromBothSides = !isSMD
```

Current behaviour with a Stepdown component:

Yes (highlighted):
[image]

No (after switching sides):
[image]

dzarda, 2021-06-07:
Might consider driving via YAML, not sure if wanted.

dzarda, 2021-06-07:
Consider a different highlight when viewed from bottom?

yaqwsx, 2022-03-01:
Implemented in d725536

closed 2022-03-01 by yaqwsx