yaqwsx/Pinion #18: Highlight from URL

dzarda, 2021-08-01:
A single link should be sufficient to open the generated page and highlight stuff in one go.

`https://blabladocs/pinion-generated.html#leds` -> [image]

We wanna be able to highlight mainly **a specific group**. More complex highlights are up for consideration but not crucial.

Also being able to specify the board side to select would be useful.

yaqwsx, 2021-08-02:
This is fairly easy on the generated template, however, I am wondering how it should work on embedded diagrams. Would having an API to programmatically highlight groups be a good solution for all the cases?

dzarda, 2021-08-02:
> Would having an API to programmatically highlight groups be a good solution for all the cases?

I don't know. I'd want to avoid JS when embedding - not sure how that would work with things like mkdocs where you want to encode stuff in a .md file. Probably will need a plugin to embed Pinion one way or another; but when writing docs all you have is links ...

dzarda, 2021-08-02:
Although a highlight API could be used from an mkdocs plugin doing the link->JS conversion

yaqwsx, 2021-08-04:
When we have API, we can include a default handler in the template. I will look at this.