yaqwsx/Pinion #19: PCB image blank

grego1981, 2021-08-04:
Hi and congrats on this concept! 
I'm trying to generate the diagram but it comes back as blank pngs.. 
What I did was to install pinion via pip3 and WSL (ubuntu20.04) in Windows 10, at first some libraries were missing (imagemagick and Wand) which I installed as well. Then I had to install kicad in WSL as well because it couldn't locate the pcbnew module. 
Then I ran :
pinion template     --board bench-PSU.kicad_pcb     --output bpsu.yaml 
and a yaml was generated, and then I ran:
pinion generate     --board bench-PSU.kicad_pcb     --specification bpsu.yaml     .
which only generated some warnings like the one below:
Warning: component 'PinHeader_1x02_P2.54mm_Vertical.back' for footprint 'SW5' from library 'Connector_PinHeader_2.54mm' was not found

And finally I ran:
pinion serve -b --directory .

And the page opened had the pins working hovering the mouse and clicking them but the PCB was a blank image. I tried the pngs directly with paint and indeed they were blanks.... 

What can I see to make it work?

yaqwsx, 2021-08-04:
This a probably because of incompatible versions of Inkscape and ImageMagick installed. Please see the discussion (linked to relevant issues) here: https://github.com/yaqwsx/PcbDraw/issues/48#issuecomment-821963357

grego1981, 2021-08-04:
Tried the patch you suggest and installed the suggested imagemagick version described in linked issue: https://github.com/ImageMagick/ImageMagick/issues/3564
Then the error changed:

Traceback (most recent call last):
  File "/home/grego/.local/bin/pinion", line 8, in <module>
    sys.exit(cli())
  File "/home/grego/.local/lib/python3.8/site-packages/click/core.py", line 1137, in __call__
    return self.main(*args, **kwargs)
  File "/home/grego/.local/lib/python3.8/site-packages/click/core.py", line 1062, in main
    rv = self.invoke(ctx)
  File "/home/grego/.local/lib/python3.8/site-packages/click/core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/grego/.local/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/grego/.local/lib/python3.8/site-packages/click/core.py", line 763, in invoke
    return __callback(*args, **kwargs)
  File "/home/grego/.local/lib/python3.8/site-packages/pinion/ui.py", line 83, in generate
    generate(specification=yaml.load(specification),
  File "/home/grego/.local/lib/python3.8/site-packages/pinion/generate.py", line 213, in generate
    fSource = generateImage(board.GetFileName(), outputdir / "front.png",
  File "/home/grego/.local/lib/python3.8/site-packages/pinion/generate.py", line 142, in generateImage
    svgToBitmap(svgfilename, outputfilename, dpi)
  File "/home/grego/.local/lib/python3.8/site-packages/pinion/generate.py", line 112, in svgToBitmap
    binaryBlob = image.make_blob(type)
  File "/home/grego/.local/lib/python3.8/site-packages/wand/image.py", line 9581, in make_blob
    with self.convert(format) as converted:
  File "/home/grego/.local/lib/python3.8/site-packages/wand/image.py", line 9539, in convert
    cloned = self.clone()
  File "/home/grego/.local/lib/python3.8/site-packages/wand/image.py", line 3348, in clone
    return Image(image=self)
  File "/home/grego/.local/lib/python3.8/site-packages/wand/image.py", line 9131, in __init__
    self.raise_exception()
  File "/home/grego/.local/lib/python3.8/site-packages/wand/resource.py", line 222, in raise_exception
    raise e
wand.exceptions.DrawError: non-conforming drawing primitive definition `circle' @ error/draw.c/RenderMVGContent/4470

yaqwsx, 2022-03-04:
The situation with ImageMagick being constantly broken made me rewrite the conversion handling in PcbDraw. You can install the upstream version of PcbDraw and then, it should work just fine.