yaqwsx/Pinion #22: New release required

Keuronde, 2021-08-14:
Hi, 

I try to use Pinion because I was impress by the tool. And I do like it, since I take the time to write an article about it (in French).
I lost a lot of time renaming all pin to have the same name for the same signal.
Now, I see that the code doing that is in commit ef2fcbfbe35c144a138f570179617296d10c33d7. It is a pity that this is not include yet in the release, now that you tool have been presented on some major web site.

I think that this modification alone deserve a new release. Can you make it ?

yaqwsx, 2021-08-14:
I plan to make a release after resolving #13, #14, #17, #18, and possibly #19.

If you are interested in having an upstream version, you can always install it via: 

```
pip install git+https://github.com/yaqwsx/Pinion@master
```

Keuronde, 2021-08-16:
Hi Yaqwsx,

Thanks for your answer.

I did try :
`pip3 install git+https://github.com/yaqwsx/Pinion@main`

But I got an error when running :
`pinion generate -b carte.kicad_pcb -s carte_tmpl_git.yml  web2/`

The error is : 
`FileNotFoundError: [Errno 2] No such file or directory: '/home/user/.local/lib/python3.7/site-packages/pinion/resources/pinion.js'`

Regards,

yaqwsx, 2021-08-17:
Oh, I forgot that setup.py does not build the Javascript framework automatically. I will try to address this near in the future.

yaqwsx, 2021-08-22:
I added the possibility to download a pre-built nightly package. The instructions for installation were added to the [installation section](https://yaqwsx.github.io/Pinion/installation/) of documenation.

Keuronde, 2021-09-02:
Hi yaqwsx,

I finally took the time to test it.
In the installation section, the link to the pre-built nightly package (https://yaqwsx.github.io/Pinion/installation/releases/pinion-latest.whl) is broken...
Sorry to bother you !

yaqwsx, 2021-09-04:
This is strange; it seems that GH pages do not serve Python packages. In the meantime, you can directly get it from the repository (branch gh-pages): https://github.com/yaqwsx/Pinion/blob/gh-pages/releases/latest/pinion-latest.whl

Keuronde, 2021-09-05:
Hi yaqwsx,

One more issue, when trying to install pinion-latest, I got the following error :

pip3 install pinion-latest.whl 
pinion-latest.whl is not a valid wheel filename.

I renamed it Pinion-0.1.0+22.g00654f4.dirty.dist-py3-none-any.whl and I was able to install it and run it. Works fine after that !

mfaccin, 2022-08-23:
I'm having the same issue that @Keuronde to install `pinion-latest` and renaming the file didn't solve it. do you have any clue about a workaround for it?

Keuronde, 2022-08-23:
@mfaccin 
To get the "good" name of the package, you need to open it with a zip tool, and get the name of the folder ".dist-info" and rename your whl file accordingly.
example
dist-info : **Pinion-0.1.0+33.g5f6607b.dirty**.dist-info
rename your wheel file : **Pinion-0.1.0+33.g5f6607b.dirty**.dist-py3-none-any.whl

mfaccin, 2022-08-23:
thank you @Keuronde 

just found the distro info with `pip3 debug pinion-latest.whl` to find the _dist-info_.

mfaccin, 2022-08-30:
hey guys @Keuronde @yaqwsx , sorry for bothering you again 

I tried to install the pinion-latest in a new machine and the ` 0.1.0+33.g5f6607b` didn't work anymore. could you help me how to find the correct wheel filename to install?

Keuronde, 2022-08-31:
Hey @mfaccin 
This morning, I try and got 404 when trying to download the file.
This afternoon, everything seem fine... With pip install Pinion-0.1.0+33.g5f6607b.dirty.dist-py3-none-any.whl

Keuronde, 2022-11-14:
Can you either explain me how you make your package or create this package?

Keuronde, 2022-11-15:
Hi yaqwsx,

I just try the new version and got a typeError...
I really would like to build a package, on't you have any instruction (otherwise, I will compare what's in your package and what in the package build with `python3 setup.py bdist_wheel`)

```
Successfully installed Pinion-0.1.0+34.gfcd0f5a.dirty pcbdraw-1.0.1 pcbnewTransition-0.2.0 svgpathtools-1.4.1
$ pinion generate  plotted -b Holonome2023.kicad_pcb -s Holonome.yml --pack web/
Traceback (most recent call last):
  File "/home/keuronde/.local/bin/pinion", line 8, in <module>
    sys.exit(cli())
  File "/home/keuronde/.local/lib/python3.9/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/keuronde/.local/lib/python3.9/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/keuronde/.local/lib/python3.9/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/keuronde/.local/lib/python3.9/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/keuronde/.local/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/keuronde/.local/lib/python3.9/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/keuronde/.local/lib/python3.9/site-packages/pinion/ui.py", line 72, in wrapper
    return func(*args, **kwargs)
  File "/home/keuronde/.local/lib/python3.9/site-packages/pinion/ui.py", line 105, in generatePlotted
    generate(specification=yaml.load(specification),
  File "/home/keuronde/.local/lib/python3.9/site-packages/pinion/generate.py", line 328, in generate
    fSource, bSource = imageGenerator(board, outputdir)
  File "/home/keuronde/.local/lib/python3.9/site-packages/pinion/ui.py", line 98, in generateImages
    return generateDrawnImages(board, outputdir, dpi, {
  File "/home/keuronde/.local/lib/python3.9/site-packages/pinion/generate.py", line 262, in generateDrawnImages
    fSource = generateImage(board.GetFileName(), outputdir / "front.png",
  File "/home/keuronde/.local/lib/python3.9/site-packages/pinion/generate.py", line 193, in generateImage
    image = plotter.plot()
  File "/home/keuronde/.local/lib/python3.9/site-packages/pcbdraw/plot.py", line 1039, in plot
    self._build_libs_path()
  File "/home/keuronde/.local/lib/python3.9/site-packages/pcbdraw/plot.py", line 1144, in _build_libs_path
    for l in self.libs:
TypeError: 'NoneType' object is not iterable

```

Polsaker, 2022-12-04:
Seeing the same error as above when plotting and this error when using the "rendered" option:

```
Traceback (most recent call last):
  File "/home/polsaker/.local/lib/python3.10/site-packages/pcbdraw/renderer.py", line 353, in startPcbnewSession
    yield session
  File "/home/polsaker/.local/lib/python3.10/site-packages/pcbdraw/renderer.py", line 516, in renderBoard
    with session.start3DViewer() as viewer:
  File "/usr/lib/python3.10/contextlib.py", line 135, in __enter__
    return next(self.gen)
  File "/home/polsaker/.local/lib/python3.10/site-packages/pcbdraw/renderer.py", line 203, in start3DViewer
    id = self.waitForWindow("3d viewer", 15)
  File "/home/polsaker/.local/lib/python3.10/site-packages/pcbdraw/renderer.py", line 99, in waitForWindow
    return self.waitForWindows([titlePattern], timeout, callback)
  File "/home/polsaker/.local/lib/python3.10/site-packages/pcbdraw/renderer.py", line 117, in waitForWindows
    raise TimeoutError(f"None of '{titlePatterns}' didn't appear within timeout. Available windows:\n{windows_list}")
TimeoutError: None of '['3d viewer']' didn't appear within timeout. Available windows:
- pcbnew
- Loading PCB
- PCB Editor
- KiCad PCB Editor
```

yaqwsx, 2023-01-12:
FYI: v0.3 was released.

yaqwsx, 2023-01-12:
@Polsaker: Please open a new issue for the problem with rendering and plotting (separate issues, please). Please, describe the best environment you use and the steps to reproduce.

closed 2023-01-12 by yaqwsx