yaqwsx/Pinion #23: Fetch API cannot load

cr0wg4n, 2021-08-28:
Chorme console input:
```
Fetch API cannot load file:///home/XXXXXX/Desktop/Proyecto%20Final/proyecto-final-hardware/demo/spec.json. URL scheme "file" is not supported.
```

I think you use the `spec.json` file as a file structure it's great but recently when I try this library the browsers (firefox and chrome) have the same problem, it is a normal beahviur or maybe I wrong?.

That's my `template.html` script configuration:
```
  <script>
      pinion.setup(document.getElementById("root"), {
          source: "."
      });
  </script>
```

Thanks and nice library :D.

cr0wg4n, 2021-08-28:
I read your code and I use the serve command, then all works well hahaha, the FAQ section coves that part, but I think is a relevant piece for your main explanation, thanks a lot, you help me to make a well documented project :D

yaqwsx, 2021-08-28:
I hoped that https://yaqwsx.github.io/Pinion/diagramWalkthrough/#testing-the-diagram covers it. Do you have a proposal for reasonable improvement?

cr0wg4n, 2021-08-28:
Oh well I don't read that part (I mean the testing part confuse me with testing like test files and coverage you know), that's my fault, but if I could make some improvement to avoid the browsers behaviur I suggest to embed the spec.json file inside the js file or optionally create the spec.json in a javascript file as a json variable that the pinion.js could recover it properlly througth the HTML importation, it's something ugly but it could be work without any server, with this feature the HTML site will be cross-platform (actually I need the pinion cli for serve the site as http server)

yaqwsx, 2021-08-28:
Embedding the JSON file might be an option easy to add but it does not solve the problem with cross-origin image fetching.

cr0wg4n, 2021-08-28:
The image fetching will be still working well, you're using the img HTML element for image fetching, the problem is at the spec.json file, you're using the fetch API, fetch always works well using a http server, but if you can run as a simple html.. fetch not works, embedding I think is an interesting solution, if you want expose the spec.json for no obfuscation reasons, you can use it as a javascript file and loading before pinion.js in you template.html