yaqwsx/Pinion #25: ruamel_yaml error

amad3v, 2021-09-29:
Hi,
The following command:
```
pinion generate -b sample.kicad_pcb ./
```
fails with error:
```
File "_ruamel_yaml.pyx", line 302, in _ruamel_yaml.CParser.__init__
TypeError: a string or stream input is required
```
Traceback:
```
Traceback (most recent call last):
File "/home/test/envs/pinion/bin/pinion", line 8, in <module>
sys.exit(cli())
File "/home/test/envs/pinion/lib/python3.9/site-packages/click/core.py", line 1137, in __call__
return self.main(*args, **kwargs)
File "/home/test/envs/pinion/lib/python3.9/site-packages/click/core.py", line 1062, in main
rv = self.invoke(ctx)
File "/home/test/envs/pinion/lib/python3.9/site-packages/click/core.py", line 1668, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "/home/test/envs/pinion/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
return ctx.invoke(self.callback, **ctx.params)
File "/home/test/envs/pinion/lib/python3.9/site-packages/click/core.py", line 763, in invoke
return __callback(*args, **kwargs)
File "/home/test/envs/pinion/lib/python3.9/site-packages/pinion/ui.py", line 83, in generate
generate(specification=yaml.load(specification),
File "/home/test/envs/pinion/lib/python3.9/site-packages/ruamel/yaml/main.py", line 432, in load
constructor, parser = self.get_constructor_parser(stream)
File "/home/test/envs/pinion/lib/python3.9/site-packages/ruamel/yaml/main.py", line 514, in get_constructor_parser
loader = XLoader(stream)
File "/home/test/envs/pinion/lib/python3.9/site-packages/ruamel/yaml/main.py", line 507, in __init__
CParser.__init__(selfx, stream)
File "_ruamel_yaml.pyx", line 302, in _ruamel_yaml.CParser.__init__
TypeError: a string or stream input is required
```
Environment:
```
KiCad   5.1.10

PcbDraw          0.8.0
pcbnewTransition 0.1.0
Pinion           0.2.0
PyYAML           5.4.1
ruamel.yaml      0.17.16
ruamel.yaml.clib 0.2.6
```

Any fix or workaround?

Thanks!

yaqwsx, 2021-09-30:
You simply miss `--specification <pathtospecfile>` in your command. See https://yaqwsx.github.io/Pinion/diagramWalkthrough/.

Nevertheless I have to add a proper, user-friendly error message.

yaqwsx, 2021-09-30:
Addressed in c2d3688

closed 2021-09-30 by yaqwsx