yaqwsx/Pinion #27: pinion generate throws error: pcbdraw returned non-zero exit status

Optoneer, 2022-02-23:
Hi,

I have Ubuntu 20.04 through WSL on my Windows 10 Pro, pip installed Pinion, installed Kicad 6, installed imagemagick. (in that order, if that is important)
Running the template command worked fine (then). When running the generate command, I encountered an issue I cannot resolve myself:

> $ pinion generate -b ../ATMega4809_Breakout.kicad_pcb --specification diagram.yaml publish

```
Warning: component 'PinHeader_2x03_P2.54mm_Vertical' for footprint 'J2' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'SW_Push_1P1T_NO_CK_KMR2' for footprint 'SW1' from library 'Button_Switch_SMD' was not found
Warning: component 'QFN-48-1EP_6x6mm_P0.4mm_EP4.2x4.2mm' for footprint 'U1' from library 'Package_DFN_QFN' was not foundWarning: component 'PinHeader_1x05_P2.54mm_Horizontal' for footprint 'J1' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'LED_RGB_5050-6' for footprint 'D1' from library 'LED_SMD' was not found
Warning: component 'PinHeader_2x02_P2.54mm_Vertical' for footprint 'J3' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'Crystal_DS26_D2.0mm_L6.0mm_Horizontal.back' for footprint 'Y1' from library 'Crystal' was not found
Traceback (most recent call last):
  File "/home/marius/.local/bin/pcbdraw", line 8, in <module>
    sys.exit(main())
  File "/home/marius/.local/lib/python3.8/site-packages/pcbdraw/pcbdraw.py", line 953, in main
    postprocess_svg(args.output, args.shrink)
  File "/home/marius/.local/lib/python3.8/site-packages/pcbdraw/pcbdraw.py", line 815, in postprocess_svg
    shrink_svg(svgfilepath, shrinkBorder)
  File "/home/marius/.local/lib/python3.8/site-packages/pcbdraw/pcbdraw.py", line 769, in shrink_svg
    paths = document.paths()
  File "/home/marius/.local/lib/python3.8/site-packages/svgpathtools/document.py", line 261, in paths
    return flattened_paths(self.tree.getroot(), group_filter,
  File "/home/marius/.local/lib/python3.8/site-packages/svgpathtools/document.py", line 143, in flattened_paths
    path = transform(parse_path(converter(path_elem)), path_tf)
  File "/home/marius/.local/lib/python3.8/site-packages/svgpathtools/path.py", line 309, in transform
    return transform_segments_together(curve, transformation)
  File "/home/marius/.local/lib/python3.8/site-packages/svgpathtools/path.py", line 191, in transform_segments_together
    transformed_segs = [transformation(seg) for seg in path]
  File "/home/marius/.local/lib/python3.8/site-packages/svgpathtools/path.py", line 191, in <listcomp>
    transformed_segs = [transformation(seg) for seg in path]
  File "/home/marius/.local/lib/python3.8/site-packages/svgpathtools/path.py", line 308, in <lambda>
    transformation = lambda seg: transform(seg, tf)
  File "/home/marius/.local/lib/python3.8/site-packages/svgpathtools/path.py", line 339, in transform
    return Arc(new_start, radius=new_radius, rotation=curve.rotation + rot,
  File "/home/marius/.local/lib/python3.8/site-packages/svgpathtools/path.py", line 1448, in __init__
    self._parameterize()
  File "/home/marius/.local/lib/python3.8/site-packages/svgpathtools/path.py", line 1503, in _parameterize
    raise ValueError("No such elliptic arc exists.")
ValueError: No such elliptic arc exists.
Traceback (most recent call last):
  File "/home/marius/.local/bin/pinion", line 8, in <module>
    sys.exit(cli())
  File "/home/marius/.local/lib/python3.8/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/home/marius/.local/lib/python3.8/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/home/marius/.local/lib/python3.8/site-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/marius/.local/lib/python3.8/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/marius/.local/lib/python3.8/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/home/marius/.local/lib/python3.8/site-packages/pinion/ui.py", line 83, in generate
    generate(specification=yaml.load(specification),
  File "/home/marius/.local/lib/python3.8/site-packages/pinion/generate.py", line 213, in generate
    fSource = generateImage(board.GetFileName(), outputdir / "front.png",
  File "/home/marius/.local/lib/python3.8/site-packages/pinion/generate.py", line 140, in generateImage
    subprocess.run(command, check=True)
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['pcbdraw', '--shrink', '0', '../ATMega4809_Breakout.kicad_pcb', '/tmp/tmppa6it2b7/img.svg']' returned non-zero exit status 1.
```

Is it an issue, that it does not find the libraries? I made the board under Windows, but would like to try out your tool.

What can I do?

Cheers, Marius

yaqwsx, 2022-02-23:
This is a problem in PcbDraw; it seems to not be able to handle a curve coming from the board. Could you share the board so I can reproduce the issue?

Optoneer, 2022-02-23:
Thanks for your response. With your comment I investigated the issue and started to remove items from the board. It seems to me, that vias lead to the issue. If the via diameter is less than or equal to 0.8 *pinion generate* fails. (I always ran *pinion template* before.) 

Please find two kicad_pcb files in a zip below : Minimal_ok let *pinion generate* pass, whereas Minimal_Error let it fail. The difference is the via dia of 0.8 and 0.9. 

Hope it helps. If not, I could also provide the complete board

[Minimal_examples.zip](https://github.com/yaqwsx/Pinion/files/8124334/Minimal_examples.zip)
.

Optoneer, 2022-02-23:
In the meantime I edit all vias in my board and ran *pinion template* and *pinion generate* without luck.

```
Traceback (most recent call last):                                                                             
  File "/home/marius/.local/bin/pinion", line 8, in <module>                                                   
    sys.exit(cli())                                                                                            
  File "/home/marius/.local/lib/python3.8/site-packages/click/core.py", line 1128, in __call__                 
    return self.main(*args, **kwargs)                                                                          
  File "/home/marius/.local/lib/python3.8/site-packages/click/core.py", line 1053, in main                     
    rv = self.invoke(ctx)                                                                                      
  File "/home/marius/.local/lib/python3.8/site-packages/click/core.py", line 1659, in invoke                   
    return _process_result(sub_ctx.command.invoke(sub_ctx))                                                    
  File "/home/marius/.local/lib/python3.8/site-packages/click/core.py", line 1395, in invoke                   
    return ctx.invoke(self.callback, **ctx.params)                                                             
  File "/home/marius/.local/lib/python3.8/site-packages/click/core.py", line 754, in invoke                    
    return __callback(*args, **kwargs)                                                                         
  File "/home/marius/.local/lib/python3.8/site-packages/pinion/ui.py", line 83, in generate                    
    generate(specification=yaml.load(specification),                                                           
  File "/home/marius/.local/lib/python3.8/site-packages/pinion/generate.py", line 230, in generate             
    "components": componentsDefinition(specification["components"], board),                                    
  File "/home/marius/.local/lib/python3.8/site-packages/pinion/generate.py", line 90, in componentsDefinition  
    "front": footprint.GetLayer() == pcbnew.F_Cu,                                                              
AttributeError: 'NoneType' object has no attribute 'GetLayer'                                                  
```

Here is the original board as zip file.
[ATMega4809_Breakout.zip](https://github.com/yaqwsx/Pinion/files/8125685/ATMega4809_Breakout.zip)

Sampozzo, 2022-02-28:
Ihave the same exact problem and it is seems not to be a pcbdraw issue because i can generate images with pcbdraw from the same pcb without error.

I solved reading this post, i had to set all the vias diameter to 1,2mm and via holes to 1,00mm.

Another problem i had is that every pint need oval pad shape and NOT circular. and f.paste and b.paste deselected. I had to modify every pin in my project and now it works. See the pictures
[image]
[image]

yaqwsx, 2022-03-01:
I just tested your board and it works fine for me. Could you give me output of the following commands? 
- `pip show pinion`
- `pip show pcbdraw`
- `pip show svgpathtools`
There will be probably an outdated dependency somewhere. The output of the commands will help me to track it.

Optoneer, 2022-03-01:
Thanks for testing and tracking. Here you are.

marius@Amd8Core:/mnt/c/Users/Marius$ **pip show pinion**
```
Name: Pinion
Version: 0.2.0
Summary: Create interactive pinout diagrams for KiCAD PCBs
Home-page: https://github.com/yaqwsx/Pinion
Author: Jan Mrázek
Author-email: email@honzamrazek.cz
License: UNKNOWN
Location: /home/marius/.local/lib/python3.8/site-packages
Requires: ruamel.yaml, pcbdraw, pcbnewTransition, click
Required-by:
```

marius@Amd8Core:/mnt/c/Users/Marius$ **pip show pcbdraw**
```
Name: PcbDraw
Version: 0.9.0
Summary: Utility to produce nice looking drawings of KiCAD boards
Home-page: https://github.com/yaqwsx/PcbDraw
Author: Jan Mrázek
Author-email: email@honzamrazek.cz
License: UNKNOWN
Location: /home/marius/.local/lib/python3.8/site-packages
Requires: click, svgpathtools, pcbnewTransition, pybars3, wand, lxml, mistune, numpy, pyyaml
Required-by: Pinion
```

marius@Amd8Core:/mnt/c/Users/Marius$ **pip show svgpathtools**
```
Name: svgpathtools
Version: 1.4.4
Summary: A collection of tools for manipulating and analyzing SVG Path objects and Bezier curves.
Home-page: https://github.com/mathandy/svgpathtools
Author: Andy Port
Author-email: AndyAPort@gmail.com
License: MIT
Location: /home/marius/.local/lib/python3.8/site-packages
Requires: numpy, svgwrite, scipy
Required-by: PcbDraw
```

yaqwsx, 2022-03-02:
This is indeed caused by a regression in `svgpathtools`. Installing version 1.4.1 helps until we solve this in upstream; see relevant issue https://github.com/mathandy/svgpathtools/issues/171

Optoneer, 2022-03-02:
That solves the issue of "not generating", thanks. Could you tell me, what it means, when it does not find any of the components? Kicad PCB editor and the 3D viewer do display everything as expected (some 3D parts are removed).

**pinion generate --pack -b ./ATMega4809_Breakout.kicad_pcb -s ./diagram.yaml ./publish**
```
Warning: component 'PinHeader_2x03_P2.54mm_Vertical' for footprint 'J2' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'SW_Push_1P1T_NO_CK_KMR2' for footprint 'SW1' from library 'Button_Switch_SMD' was not found
Warning: component 'QFN-48-1EP_6x6mm_P0.4mm_EP4.2x4.2mm' for footprint 'U1' from library 'Package_DFN_QFN' was not found
Warning: component 'PinHeader_1x05_P2.54mm_Horizontal' for footprint 'J1' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'LED_RGB_5050-6' for footprint 'D1' from library 'LED_SMD' was not found
Warning: component 'PinHeader_2x02_P2.54mm_Vertical' for footprint 'J3' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'Crystal_DS26_D2.0mm_L6.0mm_Horizontal.back' for footprint 'Y1' from library 'Crystal' was not found
Warning: component 'Crystal_DS26_D2.0mm_L6.0mm_Horizontal' for footprint 'Y1' from library 'Crystal' was not found
Warning: component 'PinHeader_2x03_P2.54mm_Vertical.back' for footprint 'J2' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'SW_Push_1P1T_NO_CK_KMR2.back' for footprint 'SW1' from library 'Button_Switch_SMD' was not found
Warning: component 'QFN-48-1EP_6x6mm_P0.4mm_EP4.2x4.2mm.back' for footprint 'U1' from library 'Package_DFN_QFN' was not found
Warning: component 'PinHeader_1x05_P2.54mm_Horizontal.back' for footprint 'J1' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'LED_RGB_5050-6.back' for footprint 'D1' from library 'LED_SMD' was not found
Warning: component 'PinHeader_1x16_P2.54mm_Vertical.back' for footprint 'J5' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'PinHeader_2x02_P2.54mm_Vertical.back' for footprint 'J3' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'PinHeader_1x16_P2.54mm_Vertical.back' for footprint 'J4' from library 'Connector_PinHeader_2.54mm' was not found
```

**pinion serve -b --directory ./publish/** shows a "blank" board. Only when marking, something is "visible":

[image]

yaqwsx, 2022-03-04:
The situation with ImageMagick being constantly broken made me rewrite the conversion handling in PcbDraw. You can install the upstream version of PcbDraw and then, it should work just fine.

Optoneer, 2022-03-04:
I installed the upstream version by  **pip install git+https://github.com/yaqwsx/PcbDraw@master** .

So, **pip show pcbdraw** gives
```
Name: PcbDraw
Version: 0.9.0+12.g56706d6
Summary: Utility to produce nice looking drawings of KiCAD boards
Home-page: https://github.com/yaqwsx/PcbDraw
Author: Jan Mrázek
Author-email: email@honzamrazek.cz
License: UNKNOWN
Location: /home/marius/.local/lib/python3.8/site-packages
Requires: lxml, svgpathtools, pybars3, mistune, engineering-notation, numpy, pyyaml, pcbnewTransition
Required-by: Pinion
```

The hash seems to be the lastest commit from today morning.

Please also find the installed imagemagick version: **apt show imagemagick**

```
Package: imagemagick
Version: 8:6.9.10.23+dfsg-2.1ubuntu11.4
Priority: optional
Section: universe/graphics
Origin: Ubuntu
(...)
```

Then I tried the simpler test: **pcbdraw ATMega4809_breakout.kicad_pcb output.svg** and it fails with all these warnings.

Could it be an issue, that the board-file is made under windows and the kicad in WSL is not able to correctly process the correct paths? However, I tried to replace the paths and env.variables linking to certain libraries. I also tried to export all used footprints into a single (local) library, unfortunately without success. It shows up even more warnings.
It's hard to me to dig into any path issues with kicad under Ubuntu (in WSL) ... I am not that familar with the Kicad config and where Kicad saves any data...

Let me know, if I can check/test/do anything else to investigate the issue.

yaqwsx, 2022-03-04:
> Then I tried the simpler test: pcbdraw ATMega4809_breakout.kicad_pcb output.svg and it fails with all these warnings.

Which warnings? Could you show them to me? The change in PcbDraw is such that it no longer uses the unreliable ImageMagick, but instead, it uses either Inkscape or rsvg-convert (at least one of them has to be installed).

Optoneer, 2022-03-04:
I have installed inkscape now (sudo apt install inkscape) and installed rsvg-convert (sudo apt install librsvg2-bin) and get for 
**pcbdraw ATMega4809_Breakout.kicad_pcb output.svg**

```
Warning: component 'PinHeader_2x03_P2.54mm_Vertical' for footprint 'J2' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'SW_Push_1P1T_NO_CK_KMR2' for footprint 'SW1' from library 'Button_Switch_SMD' was not found
Warning: component 'QFN-48-1EP_6x6mm_P0.4mm_EP4.2x4.2mm' for footprint 'U1' from library 'Package_DFN_QFN' was not found
Warning: component 'PinHeader_1x05_P2.54mm_Horizontal' for footprint 'J1' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'LED_RGB_5050-6' for footprint 'D1' from library 'LED_SMD' was not found
Warning: component 'PinHeader_2x02_P2.54mm_Vertical' for footprint 'J3' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'Crystal_DS26_D2.0mm_L6.0mm_Horizontal.back' for footprint 'Y1' from library 'Crystal' was not found
```

yaqwsx, 2022-03-04:
My bad; I haven't pushed all the changes. Please update both, PcbDraw and Pinion to upstream version and give it a try.

Optoneer, 2022-03-04:
Sorry ... I installed the latest upstreams, ran the command and got warnings. Please find the versions below, I checked the hashs and they are the same as shown on github. 

**pcbdraw ATMega4809_Breakout.kicad_pcb output.svg**
```
Warning: component 'PinHeader_2x03_P2.54mm_Vertical' for footprint 'J2' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'SW_Push_1P1T_NO_CK_KMR2' for footprint 'SW1' from library 'Button_Switch_SMD' was not found
Warning: component 'QFN-48-1EP_6x6mm_P0.4mm_EP4.2x4.2mm' for footprint 'U1' from library 'Package_DFN_QFN' was not found
Warning: component 'PinHeader_1x05_P2.54mm_Horizontal' for footprint 'J1' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'LED_RGB_5050-6' for footprint 'D1' from library 'LED_SMD' was not found
Warning: component 'PinHeader_2x02_P2.54mm_Vertical' for footprint 'J3' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'Crystal_DS26_D2.0mm_L6.0mm_Horizontal.back' for footprint 'Y1' from library 'Crystal' was not found
```

**pip show pinion**
```
Name: Pinion
Version: 0.1.0+28.gf3779ac
Summary: Create interactive pinout diagrams for KiCAD PCBs
Home-page: https://github.com/yaqwsx/Pinion
Author: Jan Mrázek
Author-email: email@honzamrazek.cz
License: UNKNOWN
Location: /home/marius/.local/lib/python3.8/site-packages
Requires: pcbdraw, pcbnewTransition, click, ruamel.yaml
Required-by:
```

**pip show drawpcb**
```
WARNING: Package(s) not found: drawpcb
marius@Amd8Core:/mnt/c/temp/piniontest$ pip show pcbdraw
Name: PcbDraw
Version: 0.9.0+14.gff758ea
Summary: Utility to produce nice looking drawings of KiCAD boards
Home-page: https://github.com/yaqwsx/PcbDraw
Author: Jan Mrázek
Author-email: email@honzamrazek.cz
License: UNKNOWN
Location: /home/marius/.local/lib/python3.8/site-packages
Requires: pybars3, lxml, pcbnewTransition, engineering-notation, pyyaml, mistune, svgpathtools, numpy
Required-by: Pinion
```

yaqwsx, 2022-03-05:
The warning only says that there are some components that cannot be drawn. But is the board properly rendered on the generated page? If not, could you share the source files and generated directory by Pinion?

Optoneer, 2022-03-05:
Oh yeah, sorry for not checking the output. So, just running drawpcb gives me a svg output:

[image]

When running
**pinion template -b ./ATMega4809_Breakout.kicad_pcb -o ./diagram.yaml**
**pinion generate --pack -b ./ATMega4809_Breakout.kicad_pcb -s ./diagram.yaml ./publish** (no matter if with or without *--pack*)
I get an error beside the warnings:

```
Warning: component 'PinHeader_2x03_P2.54mm_Vertical' for footprint 'J2' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'SW_Push_1P1T_NO_CK_KMR2' for footprint 'SW1' from library 'Button_Switch_SMD' was not found
Warning: component 'QFN-48-1EP_6x6mm_P0.4mm_EP4.2x4.2mm' for footprint 'U1' from library 'Package_DFN_QFN' was not found
Warning: component 'PinHeader_1x05_P2.54mm_Horizontal' for footprint 'J1' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'LED_RGB_5050-6' for footprint 'D1' from library 'LED_SMD' was not found
Warning: component 'PinHeader_2x02_P2.54mm_Vertical' for footprint 'J3' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'Crystal_DS26_D2.0mm_L6.0mm_Horizontal.back' for footprint 'Y1' from library 'Crystal' was not found
Warning: component 'Crystal_DS26_D2.0mm_L6.0mm_Horizontal' for footprint 'Y1' from library 'Crystal' was not found
Warning: component 'PinHeader_2x03_P2.54mm_Vertical.back' for footprint 'J2' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'SW_Push_1P1T_NO_CK_KMR2.back' for footprint 'SW1' from library 'Button_Switch_SMD' was not found
Warning: component 'QFN-48-1EP_6x6mm_P0.4mm_EP4.2x4.2mm.back' for footprint 'U1' from library 'Package_DFN_QFN' was not found
Warning: component 'PinHeader_1x05_P2.54mm_Horizontal.back' for footprint 'J1' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'LED_RGB_5050-6.back' for footprint 'D1' from library 'LED_SMD' was not found
Warning: component 'PinHeader_1x16_P2.54mm_Vertical.back' for footprint 'J5' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'PinHeader_2x02_P2.54mm_Vertical.back' for footprint 'J3' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'PinHeader_1x16_P2.54mm_Vertical.back' for footprint 'J4' from library 'Connector_PinHeader_2.54mm' was not found
Traceback (most recent call last):
  File "/home/marius/.local/bin/pinion", line 8, in <module>
    sys.exit(cli())
  File "/home/marius/.local/lib/python3.8/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/home/marius/.local/lib/python3.8/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/home/marius/.local/lib/python3.8/site-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/marius/.local/lib/python3.8/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/marius/.local/lib/python3.8/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/home/marius/.local/lib/python3.8/site-packages/pinion/ui.py", line 83, in generate
    generate(specification=yaml.load(specification),
  File "/home/marius/.local/lib/python3.8/site-packages/pinion/generate.py", line 219, in generate
    packPinion(outputdir)
  File "/home/marius/.local/lib/python3.8/site-packages/pinion/generate.py", line 181, in packPinion
    get("js", f)
  File "/home/marius/.local/lib/python3.8/site-packages/pinion/get.py", line 14, in get
    with open(os.path.join(RESOURCES, mapping[what])) as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/marius/.local/lib/python3.8/site-packages/pinion/resources/pinion.js'
```

mfaccin, 2022-08-17:
did you find the solution for the _not found_ components? I'm facing the same issue with standard components from the standard 3d library @Optoneer

mfaccin, 2022-08-18:
that's awesome @yaqwsx, thank you for the feedback! does the _won't be available on Windows_ includes the WSL2 approach? :)

if I got it right the KiCad rendering preview will use the same library the 3D render of Pcbnew defines and that's great.

yaqwsx, 2022-08-18:
It will work on WSL and Docker on Windows. It just cannot do it when run natively. Unfortunately, it seems that KiCAD 7 will still miss the essential API parts that would allow us to run it natively and work faster.

yaqwsx, 2022-11-15:
I consider this resolved. We still need to work on the component library, but that is matter of PcbDraw, not Pinion.

closed 2022-11-15 by yaqwsx