yaqwsx/Pinion #34: Unknown layer ID -1 error when trying to generate pinion diagram

f5AFfMhv, 2023-01-02:
Hi, I get this error when trying following steps:

1. pinion template --board keypad.kicad_pcb --output keypad_pinion_spec.yaml
2. pinion generate rendered --board keypad.kicad_pcb --specification keypad_pinion_spec.yaml

tried 2 different projects and both rendered and plotted options.

Full error
```bash
/builddir/build/BUILD/kicad-6.0.10/common/layer_id.cpp(180): assert ""false"" failed in LayerName(): Unknown layer ID -1
Traceback (most recent call last):
  File "/home/mj/.local/bin/pcbdraw", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/mj/.local/lib/python3.11/site-packages/pcbdraw/pcbdraw.py", line 938, in main
    walk_components(board, args.back, lambda lib, name, val, ref, pos:
  File "/home/mj/.local/lib/python3.11/site-packages/pcbdraw/pcbdraw.py", line 515, in walk_components
    orient = math.radians(module.GetOrientation() / 10)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: must be real number, not EDA_ANGLE
Traceback (most recent call last):
  File "/home/mj/.local/bin/pinion", line 33, in <module>
    sys.exit(load_entry_point('Pinion==0.2.0', 'console_scripts', 'pinion')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mj/.local/lib/python3.11/site-packages/pinion/ui.py", line 83, in generate
    generate(specification=yaml.load(specification),
  File "/home/mj/.local/lib/python3.11/site-packages/pinion/generate.py", line 213, in generate
    fSource = generateImage(board.GetFileName(), outputdir / "front.png",
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mj/.local/lib/python3.11/site-packages/pinion/generate.py", line 140, in generateImage
    subprocess.run(command, check=True)
  File "/usr/lib64/python3.11/subprocess.py", line 569, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['pcbdraw', '--shrink', '0', 'keypad.kicad_pcb', '/tmp/tmpylq492qm/img.svg']' returned non-zero exit status
```
Versions:
OS: Fedora 37
KiCAD Platform: Linux 6.0.12-300.fc37.x86_64 x86_64, 64 bit, Little endian, wxGTK, gnome-xorg, x11
Python 3.11.0
pinion, version 0.2.0
PcbDraw 0.9.0

yaqwsx, 2023-01-02:
I believe that you managed to install `pcbnewTransition` version `0.3`, meanwhile, the current version of Pinion requires version `0.2`

f5AFfMhv, 2023-01-02:
I downgraded `pcbnewTransition` to 0.2 and now facing different error.
It seems that I have installed bunch of modules with unsupported versions. I had to downgrade `pcbdraw` as well to version 0.9 previously. Not sure how that happened as I used `pip install pinion`.

Now I get this error:
```bash
(process:64580): Pango-WARNING **: 20:07:34.842: failed to create cairo scaled font, expect ugly output. the offending font is 'Liberation Serif 2097151.9990234375'

(process:64580): Pango-WARNING **: 20:07:34.842: font_face status is: error occurred in libfreetype

(process:64580): Pango-WARNING **: 20:07:34.842: scaled_font status is: error occurred in libfreetype
thread '<unnamed>' panicked at 'assertion failed: h >= 0.0', src/text.rs:360:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
fatal runtime error: failed to initiate panic, error 5
zsh: IOT instruction (core dumped)  pinion generate plotted --board keypad.kicad_pcb --specification
```

yaqwsx, 2023-01-02:
I don't know what system do you run, but it seems that your `libfreetype` crashes. Probably in `librsvg` that converts SVG into bitmap.

f5AFfMhv, 2023-01-02:
Tried reinstalling pinion from git repository, now I have these versions:
```bash
PcbDraw            1.0.1
pcbnewTransition   0.2.0
Pinion             0.1.0+34.gfcd0f5a
```

and I get different error yet again
```python
Traceback (most recent call last):
  File "/home/mj/.local/bin/pinion", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/usr/lib/python3.11/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mj/.local/lib/python3.11/site-packages/pinion/ui.py", line 72, in wrapper
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/mj/.local/lib/python3.11/site-packages/pinion/ui.py", line 105, in generatePlotted
    generate(specification=yaml.load(specification),
  File "/home/mj/.local/lib/python3.11/site-packages/pinion/generate.py", line 328, in generate
    fSource, bSource = imageGenerator(board, outputdir)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mj/.local/lib/python3.11/site-packages/pinion/ui.py", line 98, in generateImages
    return generateDrawnImages(board, outputdir, dpi, {
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mj/.local/lib/python3.11/site-packages/pinion/generate.py", line 262, in generateDrawnImages
    fSource = generateImage(board.GetFileName(), outputdir / "front.png",
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mj/.local/lib/python3.11/site-packages/pinion/generate.py", line 193, in generateImage
    image = plotter.plot()
            ^^^^^^^^^^^^^^
  File "/home/mj/.local/lib/python3.11/site-packages/pcbdraw/plot.py", line 1039, in plot
    self._build_libs_path()
  File "/home/mj/.local/lib/python3.11/site-packages/pcbdraw/plot.py", line 1144, in _build_libs_path
    for l in self.libs:
TypeError: 'NoneType' object is not iterable
```

I also tried older versions of `PcbDraw` with no luck. I run fedora 37 kernel 6.0.12-300